adafruit/Adafruit_LED_Backpack #20: begin() turns on display before initializing ram causing display to "glitch"

bperrybap, 2015-03-25:
Currently begin() does not set the RAM in the chip to a known state by clearing it before enabling the display.
The chip powers up with bits set in its RAM so until the user clears it, verious leds will be on.
Even if the user does this immediately after calling begin(), there will be momentary flash of leds.
To fix this, the display RAM needs to be cleared before enabling the display in begin().
i..e like this:
```
void Adafruit_LEDBackpack::begin(uint8_t _addr = 0x70) {
  i2c_addr = _addr;

  Wire.begin();

  Wire.beginTransmission(i2c_addr);
  Wire.write(0x21);  // turn on oscillator
  Wire.endTransmission();

  // ensure RAM is cleared before turning on display
  clear();
  writeDisplay();

  blinkRate(HT16K33_BLINK_OFF); // enable display, turn off blink

  setBrightness(15); // max brightness
}
```

bperrybap, 2015-06-13:
Guys, it's been a while and this seems to not be assigned yet. 
It is a an easy fix.

bperrybap, 2015-12-09:
Guys, this is going on 9 months.
It is literally a 1 line change to the code to fix this.
The clear() call needs to happen before you call writeDisplay() to ensure that the display doesn't show random CRAP during initialization.

ericandrewlewis, 2016-01-01:
Hi @bperrybap, what's the best way to see this glitch? Can you provide a code example and describe what to look for? I'd be happy to reproduce and review your suggestion.

bperrybap, 2016-01-01:
I fully described the bug in the first post. Look at the current code for begin() in the library.
The two lines of code (3 with comment) I provided fixes the initialization bug.
The current begin() code turns on the display when it calls blinkRate() without first clearing out the display memory. Any 1 bits in RAM cause LEDs to turn on.
Since the chip does not clear the display RAM on powerup and the display RAM is not being cleared by begin(), and the display RAM does not power up with all bits cleared, random LEDs will be on when begin() returns since the RAM is never cleared.

The chip powers up with the display disabled - which means all LEDs will be off.
To ensure that all the LEDs remain off, begin() should ensure that the display RAM is always cleared  before it turns on the display to ensure that you do not get any momentary LED glitches/flashes. In order to do that it must clear the display ram before it turns on the display which is turned by the call to blinkRate().
The two calls I added do this.

This glitch can be seen on all examples that come with the library. 
Just look at the leds after a powerup for any example.
Look closely and you can momentarily see random LEDs flash when being powered up as the example code is starting up.
It is pretty obvious and very annoying especially when done in a dim environment.
If you really want to see the problem for a longer period of time.
Just add a while(1);  or a delay(3000); after the begin() call and the power cycle the device.
You will see the random LEDs stuck on since the library is not properly initializing display ram in begin()

ericandrewlewis, 2016-01-01:
Thanks @bperrybap, I see the glitch now. 

To reproduce, upload [this code](https://gist.github.com/ericandrewlewis/fd92f7acddf1fe7712df) to an Arduino. Then unplug and replug the Arduino. You'll notice random segments colored on start. [Like this](https://www.youtube.com/watch?v=kjKuG1dSbDc&feature=youtu.be).

With the code change @bperrybap suggests, I still see a glitch — the left-side segments of the left-most character light up for a moment. 

If I move the clearing logic like this, I see no "glitch" at all:

```cpp
void Adafruit_LEDBackpack::begin(uint8_t _addr = 0x70) {
  i2c_addr = _addr;

  Wire.begin();

  // Clear the display buffer before turning on the display.
  clear();
  writeDisplay();

  Wire.beginTransmission(i2c_addr);
  Wire.write(0x21);  // turn on oscillator
  Wire.endTransmission();
  blinkRate(HT16K33_BLINK_OFF);

  setBrightness(15); // max brightness
}
```

@bperrybap, does this change make sense?

bperrybap, 2016-01-01:
Interesting. That is surprising - not what I would expect.
It sounds like the change I proposed didn't fully fix it for your setup?
But the video you posted is for the original code and not using the code I posted?
Are you saying that with the code update I posted, that there are no stray leds on during the 3 second delay in the sketch but that there is very short momentary LED glitch prior to the 3 second delay in the sketch?

When I did testing, I was using the part in matrix mode vs LED mode and I only had 13 of the 16 columns hooked up. Maybe there is something odd going on with those other columns since there is some special capabilities on the K15/int pin. Unfortunately I don't have access to the h/w I was using - it is part of a WordClock that is 650 miles away from me right now.
Or access to any other ht16k33 h/w at the moment.

I would have originally done it the way you have it right now other than I thought that you couldn't access the internal display ram until the oscillator was turned on.
The data sheet says the part powers up with the oscillator powered off, which is standby mode.
It also says you can't write to display ram when the part is in standby mode.
So doing the writeDisplay() before turning on the oscillator violates the datasheet spec since it says the only thing you can do when in standby mode is write to the system setup resister.

I'm wondering if there is a powerup timing issue.
The datasheet on page 11, section "Function Description" in the subsection "Power-on Reset" specifies to avoid i2c data transfers for 1ms after powerup.
Powerup timing is tricky on the Arduino boards as the AVR can start running a few ms before full voltage is achieved. I have run into this in the past. Depending on the sketch code and boot-loader being used, the timing from power being applied to the Arduino board to the AVR  running setup() and eventually calling into a library begin() routine can vary quite a bit and in some cases it can be called before the other h/w devices are operational since they need a higher VCC voltage before they start their initialization and the AVR is already off and running before the VCC is up to full voltage.
To fully satisfy this h/w spec, an additional delay at the top of begin() might be needed.

I'm wondering if moving the clear() and writeDisplay() above turning on the oscillator is really just adding needed delay vs actually clearing the display ram.

It would be useful to track down what the real issue is.
I wish I had some h/w to be able to do some testing.

I'd suggest running some tests to isolate the issue.
Here is what I'd suggest:
- test for a power up timing issue
This can be done by going back to the original/existing library begin() code and adding a delay() at the top of begin() (this is code with no clear() or writeDisplay())
For this test I'd use something large like 20ms or maybe even 50ms to ensure that this can be ruled out.

If the problem is still there, See if turning on the oscillator is the issue.
Remove the delay from the sketch and add a delay(3000); just after the endTransmission() in begin()

If the stray LEDs show up during this long delay() in begin() after endTransmission() then the issue is when the oscillator is turned on.
Not sure what to do in this case, since the datasheet says you are not allowed to talk to display ram when the part is in standby mode which it will be prior to turning on the oscillator.
And the datasheet also says the LEDs should all be off when the chip powers up so at power up, even if the RAM was garbage, no LEDs should turn on when the oscillator is enabled, since the display is disabled/off by default.

If that wasn't the issue, I'd then move the delay(3000); to after the blinkRate() call.
This is where I'd expect to see the issue with stray LEDs on.

Then it would be interesting to see if the issue goes away when there is a delay() at the top of begin() and the clear() and writeDisplay() are *after* the oscillator is enabled and just above the blinkRate() call.
I would expect this to work.
If it doesn't then I'm not sure what the issue is and perhaps there is an issue with the datasheet  with respect to how the part really works.

I'm very interested in the results of these tests.

bperrybap, 2018-04-07:
3 years and the code still has not been updated to fix this?
The issue is that the display RAM is not being initialized or cleared out in begin() even though it turns on the display.
i.e. the display RAM has garbage in it at powerup and the begin() code is not initializing it, so garbage pixels show up when begin() returns.

The display ram needs to be cleared before the display is turned on.
i.e. before blinkRate() is called.

bperrybap, 2019-03-29:
Guys, Another year has passed and this bug is still not fixed.
I provided a pull request to fix this a year ago. What is the hold up on getting this fixed?
Surely this bug and the code to fix it, given its simplicity, can not be controversial.

bperrybap, 2020-08-08:
Another year has passed and this bug is still not fixed.
I have provided a pull request to fix this issue.
What is the hold up?

closed 2020-08-08 by ladyada