adafruit/Adafruit_LED_Backpack #35: ATTINY85

TiagoAlmeida3, 2017-03-01:
Hi, 
im trying to put a 8x8 matrix (HT16K33) working with an attiny85 chip and Sparkfun Tiny Programmer but it wont work. 
My question is if you have the matrix tested for attiny85 chip on a Sparkfun Tiny Programmer  or some kind of toturial, i have seen on the library that it was tested for the chip but i can put it all working together.

nodoubtman, 2020-12-22:
Yeah tried with 7 seven sketche: 
C:\Users\MARC\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src\utility\twi.c:530:34: error: 'TWSTA' undeclared (first use in this function); did you mean 'TWS3'?

          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;

                                  ^

C:\Users\MARC\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\libraries\Wire\src\utility\twi.c:530:46: error: 'TWEN' undeclared (first use in this function); did you mean 'TWINT'?

          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;

can you help?
THanks.
Marc.

ladyada, 2020-12-23:
that error is in the Arduino core has nothing to do with this library.

closed 2020-12-23 by ladyada