adafruit/Adafruit_LED_Backpack #46: Second 14-segment HT16K33 Backpack flickers

SteveRMann, 2019-05-01:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

The problem is the same on the NodeCU and WeMos D1R1

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **Wemos Mini D1 R1**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.9**

I have a project that is using the Adafruit HT16K33 14-segment backpack.  It is working well.  However, I wanted to add a second Adafruit HT16K33 14-segment backpack.  Sounds simple enough, just change the I2C address of the second display ans make a second instance of the Adafruit_AlphaNum4 object.  And,, it works.  But there is an annoying flicker on the second display.

Rather than submit my project which is a few hundred lines of code, I tested the problem on the Adafruit example sketch,  quadalphanum.ino.  Same problem.  The second display flickers.

Here is the modified code:

```
// Demo the quad alphanumeric display LED backpack kit
// scrolls through every character, then scrolls Serial
// input onto the display

#include <Wire.h>
#include <Adafruit_GFX.h>
#include "Adafruit_LEDBackpack.h"

Adafruit_AlphaNum4 display1 = Adafruit_AlphaNum4();
Adafruit_AlphaNum4 display2 = Adafruit_AlphaNum4();

void setup() {
  Serial.begin(115200);

  display1.begin(0x70);  // pass in the address
  display2.begin(0x71);  // pass in the address

  // ****** Display1 ******
  display1.writeDigitRaw(3, 0x0);
  display1.writeDigitRaw(0, 0xFFFF);
  display1.writeDisplay();
  delay(200);
  display1.writeDigitRaw(0, 0x0);
  display1.writeDigitRaw(1, 0xFFFF);
  display1.writeDisplay();
  delay(200);
  display1.writeDigitRaw(1, 0x0);
  display1.writeDigitRaw(2, 0xFFFF);
  display1.writeDisplay();
  delay(200);
  display1.writeDigitRaw(2, 0x0);
  display1.writeDigitRaw(3, 0xFFFF);
  display1.writeDisplay();
  delay(200);

  // ****** Display2 ******
  display2.writeDigitRaw(3, 0x0);
  display2.writeDigitRaw(0, 0xFFFF);
  display2.writeDisplay();
  delay(200);
  display2.writeDigitRaw(0, 0x0);
  display2.writeDigitRaw(1, 0xFFFF);
  display2.writeDisplay();
  delay(200);
  display2.writeDigitRaw(1, 0x0);
  display2.writeDigitRaw(2, 0xFFFF);
  display2.writeDisplay();
  delay(200);
  display2.writeDigitRaw(2, 0x0);
  display2.writeDigitRaw(3, 0xFFFF);
  display2.writeDisplay();
  delay(200);

  display1.clear();
  display1.writeDisplay();
  display2.clear();
  display2.writeDisplay();

  // display every character,
  for (uint8_t i = '!'; i <= 'z'; i++) {
    display1.writeDigitAscii(0, i);
    display1.writeDigitAscii(1, i + 1);
    display1.writeDigitAscii(2, i + 2);
    display1.writeDigitAscii(3, i + 3);
    display1.writeDisplay();

    display2.writeDigitAscii(0, i);
    display2.writeDigitAscii(1, i + 1);
    display2.writeDigitAscii(2, i + 2);
    display2.writeDigitAscii(3, i + 3);
    display2.writeDisplay();

    delay(50);
  }
  Serial.println();
  Serial.println("Start typing to display!");
}

char displaybuffer[4] = {' ', ' ', ' ', ' '};

void loop() {
  while (! Serial.available()) return;

  char c = Serial.read();
  if (! isprint(c)) return; // only printable!

  // scroll down display
  displaybuffer[0] = displaybuffer[1];
  displaybuffer[1] = displaybuffer[2];
  displaybuffer[2] = displaybuffer[3];
  displaybuffer[3] = c;

  // set every digit to the buffer
  display1.writeDigitAscii(0, displaybuffer[0]);
  display1.writeDigitAscii(1, displaybuffer[1]);
  display1.writeDigitAscii(2, displaybuffer[2]);
  display1.writeDigitAscii(3, displaybuffer[3]);

  display2.writeDigitAscii(0, displaybuffer[0]);
  display2.writeDigitAscii(1, displaybuffer[1]);
  display2.writeDigitAscii(2, displaybuffer[2]);
  display2.writeDigitAscii(3, displaybuffer[3]);

  // write it out!
  display1.writeDisplay();
  display2.writeDisplay();
  delay(200);
}
```

Can this be fixed?

caternuson, 2021-09-29:
Please post this in the forums with photos of your setup.
https://forums.adafruit.com/

closed 2021-09-29 by caternuson

SteveRMann, 2021-09-30:
Sorry for the ghosting. It turns out to be a power issue. Problem solved.

caternuson, 2021-09-30:
No worries. Glad you figured it out. Thanks for responding back here with the info.