adafruit/Adafruit_LED_Backpack #53: 3 year old open PR #37 (Fix period in alphanumeric mapping) needs attention please.

grndkntrl, 2020-10-02:
Hi,

I just ran into this mistake in the library whilst trying to get the decimal-point/period/full-stop to display on a 14-segment alphanumeric LED interfaced via an HT16K33 (as used in UnexpectedMaker's W.O.P.R. https://github.com/UnexpectedMaker/wopr).

currently line 102 of Adafruit_LEDBackpack.cpp (in the alphafonttable) reads:

`0b0000000000000000, // .`

when it should be:

`0b0100000000000000, // .`

I've tested the fix and can confirm that it now displays as expected.

The PR (https://github.com/adafruit/Adafruit_LED_Backpack/pull/37) I referenced in the issue title has been open since 24/08/2017, and given that it's such a simple fix I hope it gets picked up on this time.

I've yet to check for any other characters that don't display as expected but if I find any I'll add them to this issue.

Thanks!
Glenn

nodoubtman, 2021-04-05:
Thank You So much! :)

caternuson, 2021-09-30:
Fixed with #61, which was nothing more than a re-submit of #37.

closed 2021-09-30 by caternuson