adafruit/Adafruit_LED_Backpack #68: Garbled display with Pico RP2040

caternuson, 2022-01-26:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=47&t=187543

Issue recreated locally. Getting identical wrong results, so not noise. Works fine on Itsy M4, so not 3v vs 5v. Most likely something RP2040 specific.

ladyada, 2022-01-26:
!? thats quite wierd

caternuson, 2022-01-28:
Looks potentially related to use of `pgm_read_word`. Uncommenting these debug prints:
https://github.com/adafruit/Adafruit_LED_Backpack/blob/508c0e6694d5197e598f81b5ca0337f58ddd1940/Adafruit_LEDBackpack.cpp#L338-L342
Get the following:

**Itsy M4 (expected output)**
65 / 'A' = 0xF7
66 / 'B' = 0x128F
67 / 'C' = 0x39
68 / 'D' = 0x120F

**RP2040 (garbled output)**
65 / 'A' = 0x600
66 / 'B' = 0x6
67 / 'C' = 0x2000
68 / 'D' = 0x220

I've scoped this and the I2C traffic looks fine. It's clocking out exactly what's in the buffer. So it's more an internal data type/size thing specific to RP2040.

ladyada, 2022-01-28:
sounds like pgmreadword doesnt do what it should - which is nothing?

caternuson, 2022-01-29:
Opened issue with core:
https://github.com/earlephilhower/arduino-pico/issues/447

earlephilhower, 2022-01-29:
Release 1.9.15 or arduino-pico should fix this.  The pgm_read macros support misaligned access (since the RP2040 HW does not) and having `base + offset` exposed some missing parens().

caternuson, 2022-01-29:
@earlephilhower Thanks for the quick fix! This resolved the forum issue. :)

closed 2022-01-29 by caternuson

ladyada, 2022-01-29:
@caternuson now you see why im so paranoid about parens everywhere! they'll getcha :)

ladyada, 2022-01-29:
@earlephilhower thank you for the fix!

earlephilhower, 2022-01-30:
Not a problem, it was my mistake to begin with... And I'd say the learning from this is that while parenthesis are important, `#define` macros with parameters are probably the wrong answer to whatever problem you're having!