adafruit/Adafruit_LED_Backpack #70: writeDataRaw() & writeDigitAscii() descriptions are incorrect & incomplete

ricknun, 2022-03-16:
http://adafruit.github.io/Adafruit_LED_Backpack/html/class_adafruit__7segment.html
says writeDigitRaw() parameter x is "Character position (0-3)."  It is 0-4 with character position 2 being the ":" in the display.  It would also be helpful to explain that both LEDs in the display ":" (I expected them to be individually controlled) are controlled by bit1 of the bitmask parameter (I expected bit0) and all other bitmask bits are ignored.

Similarly it says writeDigitAscii() parameter x is "Character position" and nothing more.  It is also 0 to 4 with character 2 being the ":" in the display.  It is unclear what to expect when writing an ASCII character with "dot" true or false to the display ":" (I didn't try it.)  It seems like this is an coding error that now should just be clearly documented and supported. The function interface would have been better written as character positions 0 to 3 and nothing to do with the ":" in the display.

I'm not sure if those are the only errors on that page, but encourage someone that understands the code better to proofread the Doxygen documentation -- It is the first place I go when trying to understand libraries.  And many thanks for the libraries -- They are very useful and valuable.

SleepySpaceBear, 2023-10-20:
I'm about to make a fork then PR to fix the source code so the behavior matches the documentation in the header file.

ricknun, 2023-10-20:
SleepySpaceBear: I hope you don't change the behavior of the code in any way that makes my (and others') using source code have to change -- no backward incompatibilities.  I meant to suggest the documentation change to agree with what the code is already doing, not the other way around.

SleepySpaceBear, 2023-10-20:
I'm doing both as separate PRs and letting the maintainers choose