adafruit/Adafruit_LED_Backpack #73: 1 segment consistently missing on each character with Adafruit 14-segment backpack

pomplesiegel, 2022-12-27:
Hello! In general this library is working very well for me. However, I have encountered one issue: 

For all of the characters on my Adafruit 14 segment backpack + diplay there is one segment missing. It is the right middle dash location. See image below: 

[image]

I have found that this occurs whether I am using `writeDigitNum()` or `writeDigitAscii()`. Everything displays correctly except this one segment. 

Any thoughts? 
Thank you!

Library version: 1.3.2
Device: Teensy 4.1 
Platform: "arduino" via PlatformIO

ladyada, 2022-12-30:
its probably hardware/soldering issue. for tech support please post to the forums

closed 2022-12-30 by ladyada

pomplesiegel, 2022-12-30:
Hello, 

I’ll check again when I have the hardware in front of me but I believe the segment DOES illuminate if you write a 0xFF raw value to the unit. 

Also it occurs on all of my displays, and I double checked and repeated all soldering after I noticed the issue.

Would you mind checking if you see the same issue on your unit using the same software and hardware?

Thank you,
Michael

pomplesiegel, 2022-12-31:
Hello @ladyada 

Indeed, if I write `writeDigitRaw(0,0xFF)` all of the segments successfully illuminate, as shown in the picture below. Therefore I believe this is a library issue, not a hardware problem. 

Would you mind testing to see if you can replicate the issue as well? 

Thank you! 

[image]

ladyada, 2022-12-31:
@pomplesiegel can you post non-platformio (we dont use it) arduino example sketch we can use to replicate? our test code works fine so we're thinking its maybe an edge case?

pomplesiegel, 2023-01-01:
@ladyada, thanks for the response! 

Just tried using Arduino IDE (Technically Teensyduino) with the latest version of this library library + dependencies via Arduino IDE's library interface, and it reproduces the issue. 

The issue occurs on my Teensy 4.1 + LED backpack hardware when I run the following simple code: 

````cpp
#include "Adafruit_LEDBackpack.h"

Adafruit_7segment display = Adafruit_7segment();

void setup() {
  display.begin(0x70);
}

void loop() {
  display.writeDigitNum(0, 4 ); //digit 0
  display.writeDigitNum(1, 5 ); //digit 1
  display.writeDigitNum(2, 8 ); //digit 2
  display.writeDisplay();
  delay(100);
}
````

See resulting picture of actual display output missing a segment: 

[image]

dhalbert, 2023-01-01:
Hi - You're using the 7-segment part of the library on a 14-segment display. Use the `Adafruit_AlphaNum4` class instead of the `Adafruit_7segment` class. See here: https://learn.adafruit.com/14-segment-alpha-numeric-led-featherwing/usage. In the Arduino library, there are also  `quadalphanum*` examples.

pomplesiegel, 2023-01-01:
@dhalbert , indeed that was the issue! Just changed everything to the Adafruit_AlphaNum4 class, adding 48 to my numeric values in order to get the right ASCII character, and the display is now working perfectly.

I apologize for the silly error and false alarm. Thank you all for your help!