adafruit/Adafruit_LED_Backpack #80: FOR TWO DISPLAYS

Chemita23, 2023-12-20:
This code is for two displays

```
// Demo the quad alphanumeric display LED backpack kit
// scrolls through every character, then scrolls Serial
// input onto the display

#include <Wire.h>
#include <Adafruit_GFX.h>
#include "Adafruit_LEDBackpack.h"

Adafruit_AlphaNum4 alpha4 = Adafruit_AlphaNum4();
Adafruit_AlphaNum4 alpha8 = Adafruit_AlphaNum4();

void setup() {
  Serial.begin(9600);
  
  alpha4.begin(0x71);  // pass in the address
  alpha8.begin(0x70);  // pass in the address

  alpha4.writeDigitRaw(3, 0x0);
  alpha4.writeDigitRaw(0, 0xFFFF);
  alpha4.writeDisplay();
  delay(200);
  alpha4.writeDigitRaw(0, 0x0);
  alpha4.writeDigitRaw(1, 0xFFFF);
  alpha4.writeDisplay();
  delay(200);
  alpha4.writeDigitRaw(1, 0x0);
  alpha4.writeDigitRaw(2, 0xFFFF);
  alpha4.writeDisplay();
  delay(200);
  alpha4.writeDigitRaw(2, 0x0);
  alpha4.writeDigitRaw(3, 0xFFFF);
  alpha4.writeDisplay();
  delay(200);

  alpha8.writeDigitRaw(3, 0x0);
  alpha8.writeDigitRaw(0, 0xFFFF);
  alpha8.writeDisplay();
  delay(200);
  alpha8.writeDigitRaw(0, 0x0);
  alpha8.writeDigitRaw(1, 0xFFFF);
  alpha8.writeDisplay();
  delay(200);
  alpha8.writeDigitRaw(1, 0x0);
  alpha8.writeDigitRaw(2, 0xFFFF);
  alpha8.writeDisplay();
  delay(200);
  alpha8.writeDigitRaw(2, 0x0);
  alpha8.writeDigitRaw(3, 0xFFFF);
  alpha8.writeDisplay();
  delay(200);
  
  alpha4.clear();
  alpha4.writeDisplay();
  alpha8.clear();
  alpha8.writeDisplay();  

  // display every character, 
  for (uint8_t i='!'; i<='v'; i++) {
    alpha4.writeDigitAscii(0, i);
    alpha4.writeDigitAscii(1, i+1);
    alpha4.writeDigitAscii(2, i+2);
    alpha4.writeDigitAscii(3, i+3);
    
    alpha8.writeDigitAscii(0, i+4);
    alpha8.writeDigitAscii(1, i+5);
    alpha8.writeDigitAscii(2, i+6);
    alpha8.writeDigitAscii(3, i+7);

    alpha4.writeDisplay();
    alpha8.writeDisplay();    
    delay(70);
  }
  Serial.println("Start typing to display!");
}

char displaybuffer[8] = {' ', ' ', ' ', ' ', ' ', ' ', ' ', ' '};

void loop() {
  while (! Serial.available()) return;
  
  char c = Serial.read();
  if (! isprint(c)) return; // only printable!
  
  // scroll down display
  displaybuffer[0] = displaybuffer[1];
  displaybuffer[1] = displaybuffer[2];
  displaybuffer[2] = displaybuffer[3];
  displaybuffer[3] = displaybuffer[4];
  displaybuffer[4] = displaybuffer[5];
  displaybuffer[5] = displaybuffer[6];
  displaybuffer[6] = displaybuffer[7];
  displaybuffer[7] = c;
 
  // set every digit to the buffer
  alpha4.writeDigitAscii(0, displaybuffer[0]);
  alpha4.writeDigitAscii(1, displaybuffer[1]);
  alpha4.writeDigitAscii(2, displaybuffer[2]);
  alpha4.writeDigitAscii(3, displaybuffer[3]);
  
  alpha8.writeDigitAscii(0, displaybuffer[4]);
  alpha8.writeDigitAscii(1, displaybuffer[5]);
  alpha8.writeDigitAscii(2, displaybuffer[6]);
  alpha8.writeDigitAscii(3, displaybuffer[7]);  
  
  alpha4.writeDisplay();
  alpha8.writeDisplay();  
  delay(200);
}
```

caternuson, 2024-02-06:
Is there an issue here? Or is this just sharing an example that uses two displays?

Chemita23, 2024-02-06:
It´s an example, sorry.

closed 2024-02-06 by caternuson