adafruit/Adafruit_LED_Backpack #86: problem **pre-install** sequence in pull request automator

phd0, 2024-02-06:
In pull request  sequence, the pre-install sequence does not go correctly.
**ci/actions_install.sh**: line 40: {: ```command not found```

[image]

this error did not appear in a previous sequence in November 2023. 
see https://github.com/adafruit/Adafruit_LED_Backpack/pull/78#partial-pull-merging

caternuson, 2024-02-06:
An issue for this has already been opened in the repo containing the CI script:
https://github.com/adafruit/ci-arduino/issues/178

Note it's just a warning and should generally not affect having CI pass.

tyeth, 2024-03-12:
Closing this as complete (CI should be fixed, please reopen if newly run actions experience a similar issue)

closed 2024-03-12 by tyeth