adafruit/Adafruit_LED_Backpack #90: setBrightness() leads to endless reboot

nimbusCee, 2025-01-22:
Hi all,

environment:
Arduino IDE 2.3.4
M5 ESP32 C3U
Adafruit 870 (LED 8x8)
latest version of libraries 'LED Backpack Library', 'GFX Library' and 'BusIO'

Before update from 2.3.3 to 2.3.4 and libraries mentioned above (don't know the versions used before), the setBrightness function worked well. Now, if used, the ESP ends up in endless reboot. When commented, in code, all is fine - except the brightness.

Best regards
Claus

nimbusCee, 2025-01-23:
Hi,
Hereby the requested code. It can be demonstrated even with the example provided 'Adafruit LED Backpack Library - matrix8x8' with additional two lines in the setup routine.
 
Wire.begin(I2C_SDA, I2C_SCL);
matrix.setBrightness(LOWEST);  // does fail ....

The ESP32 Board Support Package being used is the one by Espressif 'esp32' in version 3.1.1

[image]
  

`/*************************************************** 
  This is a library for our I2C LED Backpacks

  Designed specifically to work with the Adafruit LED Matrix backpacks 
  ----> http://www.adafruit.com/products/872
  ----> http://www.adafruit.com/products/871
  ----> http://www.adafruit.com/products/870

  These displays use I2C to communicate, 2 pins are required to 
  interface. There are multiple selectable I2C addresses. For backpacks
  with 2 Address Select pins: 0x70, 0x71, 0x72 or 0x73. For backpacks
  with 3 Address Select pins: 0x70 thru 0x77

  Adafruit invests time and resources providing this open source code, 
  please support Adafruit and open-source hardware by purchasing 
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.  
  BSD license, all text above must be included in any redistribution
 ****************************************************/

#include <Wire.h>
#include <Adafruit_GFX.h>
#include "Adafruit_LEDBackpack.h"

#define I2C_SDA 8
#define I2C_SCL 9
#define LOWEST 0

int loopValue = 0;

Adafruit_8x8matrix matrix = Adafruit_8x8matrix();

void setup() {
  Serial.begin(115200);
  Serial.println("8x8 LED Matrix Test");

  Wire.begin(I2C_SDA, I2C_SCL);
  // matrix.setBrightness(LOWEST);  // @param  b  Brightness: 0 (min) to 15 (max).
  
  matrix.begin(0x70);  // pass in the address
}

static const uint8_t PROGMEM
  smile_bmp[] =
  { B00111100,
    B01000010,
    B10100101,
    B10000001,
    B10100101,
    B10011001,
    B01000010,
    B00111100 },
  neutral_bmp[] =
  { B00111100,
    B01000010,
    B10100101,
    B10000001,
    B10111101,
    B10000001,
    B01000010,
    B00111100 };

void loop() {

  Serial.print("Loop ... ");
  Serial.println(loopValue++);

  matrix.clear();
  matrix.drawBitmap(0, 0, smile_bmp, 8, 8, LED_ON);
  matrix.writeDisplay();
  delay(500);

  matrix.clear();
  matrix.drawBitmap(0, 0, neutral_bmp, 8, 8, LED_ON);
  matrix.writeDisplay();
  delay(500);

}
`

caternuson, 2025-01-23:
What if you call `matrix.setBrightness()` **after** calling `matrix.begin()`? In general, `matrix.begin()` should be the first thing done.

nimbusCee, 2025-01-23:
Hi Nelson,

yeeeees, it works. Of course, it absolutely makes sense - and definitely, it was not due to the updates.

Thanks a lot for your quick support, all the best
Claus

closed 2025-01-23 by caternuson